rtic-rs/rfcs #16: Road Map

perlindgren, 2019-09-22:
Place Holder for further discussions.

The idea is that this issue would provide a list of "issues" and "rfcs" for discussing the overall direction of RTFM.

bfrog, 2019-12-04:
Docs that cover using pacs/hal crates more generally, specifically examples showing how the hal/pac is exposed in the context.

perlindgren, 2020-01-28:
> Docs that cover using pacs/hal crates more generally, specifically examples showing how the hal/pac is exposed in the context.
> 
> I created a fun example of a bouncy rust logo (has a bug with the ssd1306 driver) with rtfm that might be useful for this sort of thing. I want to write a blog post once the bug is sorted out. [jamwaffles/ssd1306#99](https://github.com/jamwaffles/ssd1306/issues/99)

Sounds great. We now have a section for examples.