rtic-rs/rfcs #9: Timer Queue without Systick

tpwrules, 2019-09-22:
I am experimenting with RTFM on the nrf52840 and am having some issues with the timer queue because of the chip's unconventional Systick implementation.

According to page 19 of the [nrf52840 Product Specification v1.1](https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.1.pdf), the Systick does not count when the CPU is stopped, i.e. during WFI, WFE, or SLEEPONEXIT. This means that the idle function has to be an empty loop if otherwise not used, and cannot go to sleep. This somewhat limits the value of the timer queue since the application cannot turn off the CPU and save power when there are no timer events ready.

I expect the timer queue to remain using systick by default, but I wonder if there is room for some kind of interface to use another timer when advantageous for a particular target. Nordic suggests using the RTC timer instead, or a TIMER. I don't know what other platforms this might apply to.

perlindgren, 2019-09-22:
In the long run we may opt for a Trait for generic timer implementation for timer queues. We might also want to allow for multiple timer queues. That would imply syntax extension and should be separate issue.

AfoHT, 2022-03-01:
With RTIC 1.0 it should be possible to implement Monotonic for RTC, Timer or other non-Systick clock.

closed 2022-03-01 by AfoHT