rtic-rs/rfcs #30: move/unique resources

perlindgren, 2020-03-03:
Following up on a discussion with @korken89, and the recent comparison done by @therealprof on various ways to share data to interrupt handlers. RTFM came out on top regarding, robustness/reliability (static guarantees), performance and code-size, but argued against due to added complexity.

The added complexity with RTFM can to some extent be explained by the asymmetric resource access pattern (tasks with highest or unique access to a resource gets a owned pointer the resource, while other tasks need to use the lock API).

One way to approach the problem is to follow the lines of #17, with the addition of explicitly moved resources. 

``` rust
#[rtfm::app(device = lm3s6965)]
const APP: () = {
    struct Resources {
        #[exclusive] 
        exclusevi: u32,
        #[move]
        moved: u32,
    }

    #[init(resources = [exclusive, moved])]
    fn init(cx: init::Context) {
        *cx.exclusive = 7;
        *cx.moved = 8;
        rtfm::pend(Interrupt::GPIOA);
        rtfm::pend(Interrupt::GPIOB);
    }

   #[task(binds = GPIOA, priority = 2, resources = [exclusive, moved])]
    fn gpioa(cx: gpioa::Context) {
        // no critical section
        *cx.resources.moved += 1;
        *cx.resources.exclusive += 1;
    }

    #[task(binds = GPIOB, priority = 2, resources = [exclusive])]
    fn gpiob(c: gpiob::Context) {
        // no critical section 
        *c.resources.exclusive += 1;
    }
};
``` 
Semantics:
- #[exclusive], there might be several tasks with the same prio accessing the resource without critical section.
- #[move], there must be only one task, similar to a task local resource, but set-up by init.

Implementation:

- Implementation complexity. Presumably low.
- Run-time OH. Zero.
- Guarantees, As always static, all is checked at compile time.

Ergonomics/Usability:
Syntax can be discussed. It should however indicate the semantics in an intuitive way. `exclusive` refers to the scheduling (you get exclusive access). `move` the resource is moved to the task.

Limitations:
This will not allow further `move` of resources at run-time (following the static nature and guarantees by RTFM). 

Alternative solutions.
RTFM already allows you to express this behavior implicitly, so doing nothing is an option. However, these new constructs might appeal to the non-RTFM:ers out there. With the current implementation, the asymmetric API can be confusing. This approach would go well with #17, letting the designer having static control over `exclusive` and `moved` resources.

korken89, 2020-03-03:
Overall I think you have captured what I was thinking about, great job!
This is a feature that will help users express their intent (moving or having a resource).

perlindgren, 2020-03-03:
Regarding the static mut for task local. 
``` rust
  fn uart0(c: uart0::Context) {
        static mut KALLE: u32 = 0;
        *KALLE += 1;
        c.resources.p.enqueue(42).unwrap();
    }
```
Is accepted, while, the below code is rejected.... spot the diff...

``` rust
#[task(binds = UART0, resources = [p])]
    fn uart0(c: uart0::Context) {
        c.resources.p.enqueue(42).unwrap();
        static mut KALLE: u32 = 0;
        *KALLE += 1;
    }
```
The above proposal would remove this problem.

perlindgren, 2022-01-22:
The overarching ideas have been implemented in RTIC 1.0.0. Please re-open if needed.

closed 2022-01-22 by perlindgren