rtic-rs/rfcs #32: #[init()] functions should provide a CriticalSection

BryanKadzban, 2020-04-01:
In my program's #[init()] function (using rtfm-0.5) I have to reconfigure a bunch of GPIO pins from the stm32f0xx_hal crate in order to pass them to HAL constructors like Spi::spi1() and I2c::i2c1().  Each of those GPIO-configuration calls requires a CriticalSection.  OK, I can run the call inside a cortex_m::interrupt::free() callback ... but the RTFM framework already disables interrupts before it calls the init() function, so the read-PRIMASK, cpsid, and test-old-PRIMASK-and-branch sequences from cortex_m::interrupt::free are unnecessary inside init().  But I can't express that (unless I unsafely create a CriticalSection of my own at the top of init()).

It would be nice if init::Context contained a CriticalSection so I could pass a reference to it into the HAL crate's gpioa::PA0::into_alternate_af4() function (or whichever one I need to call).

perlindgren, 2020-04-02:
Good point. I will bring the question to next rtfm meeting.

perlindgren, 2020-04-02:
Discussed briefly with @korken89, question is why the HAL would require a Critical Section, if you follow the idea of ownership of registers and the type state approach. Could you give an example perhaps, then it would be easier to see the problem.

TeXitoi, 2020-04-02:
stm32f0-hal ask for a critical section instead of a register borrow. Personally I don't like this interface but that how it's done for now.

BryanKadzban, 2020-04-02:
The HAL code is here:

https://github.com/stm32-rs/stm32f0xx-hal/blob/master/src/gpio.rs#L252

Changing that interface would work for me as well.  :)

BryanKadzban, 2020-10-03:
Looks like https://github.com/rtic-rs/cortex-m-rtic/pull/373 added this.  Thanks!

closed 2020-10-03 by BryanKadzban