rtic-rs/rfcs #35: [tracking issue] 0.6 push

korken89, 2020-05-12:
- [ ] 1. “Move back to symmetric lock API” together with “move/unique resources”
- [ ] 2. “Mod instead of const”
- [ ] 3. “Rename RTFM”

perlindgren, 2020-05-12:
3 in progress.

little-arhat, 2020-06-11:
so all email filters are broken; code breakage to follow... %)

korken89, 2020-06-12:
Sorry about that :)
We are updating documentation and communicating currently, so there might be a period of confusion.

little-arhat, 2020-06-12:
@korken89 no worries, thanks for you work! Was just surprised by all the notifications in my default inbox with non familiar name.

Btw, should `cortex-m-rtic` work now?

I have this in Cargo.toml:
```
[dependencies.cortex-m-rtic]
version = "0.5.3"
git = "https://github.com/rtic-rs/cortex-m-rtic"
rev = "cfd5f4785e9c2ae88e666d1fb94a3488904a87f5"
```

but `cargo update` fails:
```
cargo update
    Updating git repository `https://github.com/rust-embedded/cortex-m`
    Updating crates.io index
    Updating git repository `https://github.com/rtic-rs/cortex-m-rtic`
error: no matching package named `cortex-m-rtic` found
location searched: https://github.com/rtic-rs/cortex-m-rtic?rev=cfd5f4785e9c2ae88e666d1fb94a3488904a87f5
```

korken89, 2020-06-12:
Hi, you can remove the `git` and `rev`, and it should all work :)

AfoHT, 2020-10-06:
With https://github.com/rtic-rs/rtic-syntax/pull/30 and https://github.com/rtic-rs/cortex-m-rtic/pull/368 merged number 2 in the list: "Mod instead of const" hopefully can be considered done

AfoHT, 2022-09-14:
1.0 has been released (ages ago), closing :)

closed 2022-09-14 by AfoHT